STMicroelectronics/stm32u5xx-hal-driver #1: Warning unused parameter

agentxcz, 2021-12-07:
https://github.com/STMicroelectronics/stm32u5xx_hal_driver/blob/6a00c6e1cc756c9804b3044dcd4238f5612adff0/Src/stm32u5xx_hal_pwr.c#L834
and line 896 produces warning "unused parameter 'Item'" when extra warnings setting is enabled

#if defined (__ARM_FEATURE_CMSE) && (__ARM_FEATURE_CMSE == 3U) is false
assert is empty

Also stm32u5xx_hal_rcc.c line 1552, 2007, 2064.

Env: STM32Cube IDE 1.8.0
U5 MCU package ver 1.0.2

RKOUSTM, 2021-12-22:
Hi @agentxcz,

Thank you for your report. An internal tracker has been logged and a fix will be implemented and made available in the frame of a future release.

Thank you again for you report.

With regards,

RKOUSTM, 2021-12-22:
ST Internal Reference: 119943

RKOUSTM, 2022-03-15:
Hi @agentxcz,

Hope you're fine. Just to inform you the fix has been published in the frame of STM32CubeFWU5 v1.1.0 release.

With regards,

closed 2022-03-15 by RKOUSTM