STMicroelectronics/stm32u5xx-hal-driver #2: Asymmetric TX DMA for SPI causing a crash

ramin-raeisi, 2021-12-27:
**Describe the set-up**

 * custom board
 * OZone

**Describe the bug (skip if none)**

I am interfacing a SPI graphics controller on STM32U585 and to update the frame buffer data, I have to send a single byte of command followed by n bytes as RGB data(more than 65532 bytes), however between I have to toggle a GPIO in between, which controls command/data mode.

To accomplish this, I am sending the single command byte by using **HAL_SPI_Transmit_IT**, and eventually I will toggle the GPIO and initiate the TX DMA transmission for RGB data on HAL_SPI_TxCpltCallback and by using **HAL_SPI_Transmit_DMA**(multiple iterations as the whole data i more than 65532 bytes which exceed the single shot of DMA).

Crash happens in the following line as hspi->hdmarx is null in time of comparison due to not using RX DMA.

[https://github.com/STMicroelectronics/stm32u5xx_hal_driver/blob/6a00c6e1cc756c9804b3044dcd4238f5612adff0/Src/stm32u5xx_hal_spi.c#L2930]

**Additional context**
I have changed that comparison as follow
    /* DMA Normal Mode */
`    /* DMA Normal Mode */
    if (HAL_IS_BIT_CLR(cfg1, SPI_CFG1_TXDMAEN | SPI_CFG1_RXDMAEN) ||
        ((hspi->hdmarx) && (State != HAL_SPI_STATE_BUSY_RX) && (hspi->hdmarx->Mode != DMA_LINKEDLIST_CIRCULAR)) ||
        ((hspi->hdmatx) && (State != HAL_SPI_STATE_BUSY_TX) && (hspi->hdmatx->Mode != DMA_LINKEDLIST_CIRCULAR)))
`

but another issue I am facing now is that actually my DMA transmission will only work once and would fail in the second iteration as SPI state is not equal to HAL_SPI_STATE_READY in time of calling HAL_SPI_Transmit_DMA in HAL_SPI_TxCpltCallback. It seems that SPI state would not be resetted in the second call of HAL_SPI_TxCpltCallback which the after completion of first DMA transmission.

Apparently, the following line of code would not get called at all to close the transfer:
[https://github.com/STMicroelectronics/stm32u5xx_hal_driver/blob/6a00c6e1cc756c9804b3044dcd4238f5612adff0/Src/stm32u5xx_hal_spi.c#L2968] 

Actually after completion of first DMA transfer, SPI_CFG1_TXDMAEN is 1 so the comparison would fail and the aforementioned line would not get called.

I wonder if this is correct or an edge case of execution flow failure?

ASELSTM, 2022-01-31:
Hi @ramin-raeisi,

Regarding the first issue you have reported it has been already spotted and fixed internally.  The fix will be made available in the frame of a future release.

Concerning the second issue, the SPI state is updated to HAL_SPI_STATE_READY just after closing the transfer and clearing flags. 

https://github.com/STMicroelectronics/stm32u5xx_hal_driver/blob/6a00c6e1cc756c9804b3044dcd4238f5612adff0/Src/stm32u5xx_hal_spi.c#L2970

Thus, when calling `HAL_SPI_TxCpltCallback()` the SPI state is already set to **HAL_SPI_STATE_READY**. 

With regards,

ASELSTM, 2022-01-31:
ST Internal Reference: 121844

ASELSTM, 2022-03-11:
Hi @ramin-raeisi,

This issue has been fixed in the frame of version v1.1.0 of the stm32u5xx_hal_driver published recently. Please allow me then to close this thread.

With regards,

closed 2022-03-11 by ASELSTM