STMicroelectronics/stm32u5xx-hal-driver #5: stm32u5xx_hal_dma_ex.c doesn't build with arm-none-eabi-gcc

atsju, 2023-01-09:
**Caution**

This issue is not on your file but on the compiler.
However it has impact on your customers and it's in your interest to make ARM/GCC fix the issue. This issue can at least serve as notice of heads-up for next people that will encounter the problem.

**Describe the set-up**

 * arm-gnu-toolchain-12.2.rel1-mingw-w64-i686-arm-none-eabi.zip will segfault on compilation of stm32u5xx_hal_dma_ex.c with `-O3`

**Describe the bug (skip if none)**

Compiler segfault

**How to reproduce the bug (skip if none)**

1. Downloaded arm-gnu-toolchain-12.2.rel1-mingw-w64-i686-arm-none-eabi.zip from https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads
2. In attached zip, modify and run compilerBug.bat (will just compile 1 faulty file known to work)
3. observe segmentation fault in compiler

[minimalInc.zip](https://github.com/STMicroelectronics/stm32u5xx_hal_driver/files/10371554/minimalInc.zip)

**Additional context**

I have already reported bug on Linaro: https://bugs.linaro.org/show_bug.cgi?id=5924

**Screenshots**

```
C:\FW>"C:\GIT\FW_Toolchain\GNU Tools ARM Embedded\12.2.rel1\bin\arm-none-eabi-gcc.exe" -DSTM32U575xx=1 -IC:/FW/minimalInc -O3 -DNDEBUG -freport-bug -mcpu=cortex-m33 -mthumb -mfpu=fpv5-sp-d16 -mfloat-abi=hard --specs=nano.specs -ffunction-sections -fdata-sections -fno-builtin -Wall -Wfatal-errors -O3 -std=gnu17 -o stm32u5xx_hal_dma_ex.c.obj -c C:/FW/minimalInc/stm32u5xx_hal_dma_ex.c
during GIMPLE pass: evrp
C:/FW/minimalInc/stm32u5xx_hal_dma_ex.c: In function 'HAL_DMAEx_List_ReplaceNode_Head':
C:/FW/minimalInc/stm32u5xx_hal_dma_ex.c:4689:1: internal compiler error: Segmentation fault
 4689 | }
      | ^
Please submit a full bug report, with preprocessed source.
See <https://bugs.linaro.org/> for instructions.
The bug is not reproducible, so it is likely a hardware or OS problem.
```

atsju, 2023-01-09:
Note the build also fails with GCC 11.3.1 from https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads

TOUNSTM, 2023-12-20:
Hello @atsju,

Thank you for your contribution. As you said, this issue is not related to our drivers and we do not address this aspect in our GitHub repositories. We are happy that the issue has been resolved. Thanks again for your report. We look forward to hearing from you again.
Please allow me to close the issue.

With Regards,

closed 2023-12-20 by TOUNSTM