STMicroelectronics/stm32u5xx-hal-driver #8: Assert in stm32u5xx_hal_dma_ex.c breaks the lpbam utility

BojanGav, 2024-06-11:
I have been using the lpbam utility found here
https://github.com/STMicroelectronics/STM32CubeU5/tree/main/Utilities/lpbam

I recently upgraded my HAL library, and an assert was added to the stm32u5xx_hal_dma_ex.c module in the HAL_DMAEx_List_BuildNode function.

The lpbam utility uses this function to construct the dma linked lists and the lpbam utility does not set the dma mode and as a result it trips this assert.

[image]

TOUNSTM, 2024-06-12:
Hello @BojanGav,

Thank you for your report. We have tried to reproduce the problem you described, without success. Could you kindly provide us with additional details and, if possible, a code snippet to assist us in identifying the issue.

Best regards,

BojanGav, 2024-06-13:
Yeah no problem!

So I can give you the example that I am seeing, but I'm pretty sure this is the case for alot of the functions inside the LPBAM utility.

If you look inside of 

ADV_LPBAM_DMA_Start_SetFullQ

[image]

It calls LPBAM_DMA_FillNodeConfig to configure the dma node "dma_node_conf"

Then it passes a reference to "dma_node_conf" to HAL_DMAEx_List_BuildNode.

However inside of LPBAM_DMA_FillNodeConfig, the Mode parameter of the node is never set or initialized so its value is just going to be whatever happens to be on the stack at the moment dma_node_conf was declared.

It's possible that in your test it just happens to be 0 so everything works, but in my case its some random number

[image]

I think LPBAM_DMA_FillNodeConfig should be setting it to 0

I hope this helps.

Bojan

TOUNSTM, 2024-06-26:
ST Internal Reference: 185074

TOUNSTM, 2024-07-19:
Fixed in this [commit](https://github.com/STMicroelectronics/STM32CubeU5/commit/8581ab856c60da93b2c4e649449c1e8242914a37)

closed 2024-07-19 by TOUNSTM