STMicroelectronics/stm32u5xx-hal-driver #12: CRYP_AES_Decrypt forces key mode as CRYP_KEYMODE_NORMAL

mtamang10, 2024-09-12:
Hello,
In stm32u5xx_hal_cryp.c, CRYP_AES_Decrypt() function, it sets the AES_CR_KMOD to be CRYP_KEYMODE_NORMAL. I've set the hcryp->Init.KeyMode = CRYP_KEYMODE_SHARED, but I believe it's not being used. 

[image]

I believe it should be 'MODIFY_REG(hcryp->Instance->CR, AES_CR_KMOD, hcryp->Init.KeyMode);'

Thanks

ALABSTM, 2024-09-16:
Hi @mtamang10,

 You point has been forwarded to our development teams. Let's wait for their answer.

With regards,

ALABSTM, 2024-09-16:
ST Internal Reference: 191244

ALABSTM, 2024-09-24:
Hi @manisht-mwp,

According to our development teams, this is coded this way on purpose. Indeed, as you can see from [RM0456](https://www.st.com/resource/en/reference_manual/rm0456-stm32u5-series-armbased-32bit-mcus-stmicroelectronics.pdf), revision 5, there are two cryptographic instances: `AES` (chapter 49) and `SAES` (chapter 50).

The first instance, `AES`, is not intended to share any key. Instead, it is the second instance, `SAES, the secure one, that could. Once the `SAES` instance shared its key with the `AES`, this latter handles that key as a normal one.

From a functions point-of-view:
* `CRYP_AES_Decrypt()`, along with `HAL_CRYP_Encrypt()`, are intended to be used by the `AES` instance, thus, setting the `AES_CR_KMOD` to `CRYP_KEYMODE_NORMAL`.
* `HAL_CRYPEx_EncryptSharedKey()` and `HAL_CRYPEx_DecryptSharedKey()`, on the other hand, are intended to be used by the `SAES` instance.

I hope this makes things clearer. Do not hesitate otherwise.

With regards,

closed 2024-09-27 by ALABSTM