STMicroelectronics/stm32u5xx-hal-driver #14: SPI suspend not handled in blocking mode Receive

rickou, 2024-10-27:
Hi, 

After having trouble to receive big packet through SPI using HAL_SPI_Receive (clock @12MHz), I enabled the SPI_MASTER_RX_AUTOSUSP_ENABLE feature to avoid overflow.
But, now, i have another issue due to the fact this feature is not handled in the blocking form of Receive..

I know we should use DMA.
For whatever reason, the dev could wants to use the blocking version, which is almost not usable as is at high SPI frequencies (caused by other tasks/IRQ premption, or even slow CPU speeds)
So, it could be also good to add the SUSP bit clear in the blocking receive function too.. 
This way user is free to use DMA, IT or blocking receive functions

KRASTM, 2024-10-29:
Hello @rickou,

Thank you for the report.

In fact, this enhancement has already detected and implemented internally, and it will be available ASAP within the driver.

With regards,

KRASTM, 2024-10-29:
ST Internal Reference: 167877

KRASTM, 2024-11-26:
Fixed in commit: https://github.com/STMicroelectronics/stm32u5xx-hal-driver/commit/5f3b1924e386eb80963e1a8ec382ad08902e2a0d

closed 2024-11-26 by KRASTM