STMicroelectronics/stm32u5xx-hal-driver #16: HAL XSPI DTR read even byte count

jakubhecko, 2024-11-14:
**Describe the set-up**
Hello, 
using custom board with STM32U5F9 MCU, building with Clang/LLVM: 18.1.8.
Connected to a Macronix memory (MX25UM25645GXDI00) connected to the HSPI interface in 8line configuration with DTR+DQS.

**Describe the bug (skip if none)**

The issue is when using HAL_XSPI_Receive() function when HSPI configured to DTR mode, reading 1 byte leaves a HSPI periph. in a state (the second byte still in the DR register) effectively locking out any future HAL_XSPI read/write operations.

**How to reproduce the bug (skip if none)**

1. Setup a HSPI peripheral in an 8 line mode / octospi + DTR + DQS.
2. Use HAL_XSPI_Receive() to read just one byte (eg. a status register of external memory)
3. Then the HSPI periph. will read two bytes, because of the DTR mode, BUT HAL_XSPI_Receive() will read just the one byte, leaving the other in the HSPI_DR register, which in turn, leaves HSPI_SR FLEVEL non-zero and other flags set in the HSPI periph. Which hinders any further communication using the HAL_XSPI functions.

The unchecked read code is here:
https://github.com/STMicroelectronics/stm32u5xx-hal-driver/blob/e41e0c51f612cd27641e90e3c4f5a7e9388130ff/Src/stm32u5xx_hal_xspi.c#L1401C1-L1414C39

**Additional context**

I propose putting at the end of the appropriate receive function a check for non-empty FLEVEL and reading out any extra (unwanted) data.

```c 
    // Line 1428 of stm32u5xx_hal_xspi.c and on
     //------------------  Patch for DDR read of 1B ---------------------
      if(((hxspi->Instance->SR>>8) & 0x9F) > 0) {
        (void)*((__IO uint8_t *)data_reg);
      }
     //------ Rest of the original code of HAL_XSPI_Receive() ---------
    }
    else
    {
      status = HAL_ERROR;
      hxspi->ErrorCode = HAL_XSPI_ERROR_INVALID_SEQUENCE;
    }
  }

  return status;
}
```

**Code used for reading 1B**

```c
/-----------------------------------------------------------------------------
/// Read one 8bit register
uint8_t ReadRegister(uint32_t command)
//-----------------------------------------------------------------------------
{
    XSPI_RegularCmdTypeDef  sCommand;
    fill_standard_command(&sCommand);
    sCommand.Instruction        = command;
    sCommand.DataMode           = HAL_XSPI_DATA_8_LINES;
    sCommand.DataLength         = 1;
    sCommand.DummyCycles        = 0;
    /* Start Octal Software Polling to wait until BUSY=0 */
    if (HAL_XSPI_Command(&hhspi1, &sCommand, HAL_XSPI_TIMEOUT_DEFAULT_VALUE) != HAL_OK)  {
        _Error_Handler(__FILE__, __LINE__);
    }
    uint8_t tmp = 0x00;
    /* Read memory's Status register */
    if (HAL_XSPI_Receive(&hhspi1, &tmp, HAL_XSPI_TIMEOUT_DEFAULT_VALUE) != HAL_OK)  {
        _Error_Handler(__FILE__, __LINE__);
    }
    return tmp;
}

static void fill_standard_command(XSPI_RegularCmdTypeDef* sCommand)
//-----------------------------------------------------------------------------
{
    sCommand->OperationType         = HAL_XSPI_OPTYPE_COMMON_CFG;

    sCommand->IOSelect              = HAL_XSPI_SELECT_IO_7_0;

    // Instruction se doplni individualne
    sCommand->InstructionMode       = HAL_XSPI_INSTRUCTION_8_LINES;
    sCommand->InstructionWidth      = HAL_XSPI_INSTRUCTION_16_BITS;
    sCommand->InstructionDTRMode    = HAL_XSPI_INSTRUCTION_DTR_VALUE;

    sCommand->Address               = 0x0;
    sCommand->AddressMode           = HAL_XSPI_ADDRESS_NONE;
    sCommand->AddressWidth          = HAL_XSPI_ADDRESS_32_BITS;
    sCommand->AddressDTRMode        = HAL_XSPI_ADDRESS_DTR_ENABLE;

    sCommand->AlternateBytes        = 0;
    sCommand->AlternateBytesMode    = HAL_XSPI_ALT_BYTES_NONE;
    sCommand->AlternateBytesWidth   = HAL_XSPI_ALT_BYTES_8_BITS;
    sCommand->AlternateBytesDTRMode = HAL_XSPI_ALT_BYTES_DTR_ENABLE;

    sCommand->DataDTRMode           = HAL_XSPI_DATA_DTR_ENABLE;

    sCommand->DQSMode               = HAL_XSPI_DQS_DISABLE;
    sCommand->SIOOMode              = HAL_XSPI_SIOO_INST_EVERY_CMD;
} 
```

KRASTM, 2024-11-15:
Hello @jakubhecko,

Thank you for the report.

Regarding the issue, could you share your IOC file in order to reproduce the issue. Meanwhile, I have some comments on what you mentioned above:

- You mentioned that `connected to the HSPI interface in 8line configuration with DTR+DQS`, but the parameters in sCommand is disabled **(sCommand->DQSMode               = HAL_XSPI_DQS_DISABLE;)**

- DTR mode: (Double Transfer Rate), it doesn't mean it will double the data size, in other word, this mode increases the speed at which data is transferred not the size of data, `Then the HSPI periph. will read two bytes, because of the DTR mode`

With regards,

jakubhecko, 2024-11-18:
Hello @KRASTM, see the replies below:

> Regarding the issue, could you share your IOC file in order to reproduce the issue. Meanwhile, I have some comments on what you mentioned above:
>
> * You mentioned that `connected to the HSPI [...]

Yes, I had some issues with using DQS for setting up the 1B or 2B registers, so I use DQS only for the longer data transfers, not the memory register access.

> * DTR mode: (Double Transfer Rate), it doesn't mean it will double the data size, in other word, this mode increases the speed at which data is transferred not the size of data, `Then the HSPI periph. will read two bytes, because of the DTR mode`

I understand that in DTR mode the data is sampled on both edges of the clock line, that is likely why the periph returns 2B when trying to read just one.

See the IOC attached, please note, the clock tree might not be up-to-date.
[HSPI_DTR_5F9 UFBGA144.zip](https://github.com/user-attachments/files/17796779/HSPI_DTR_5F9.UFBGA144.zip)

KRASTM, 2024-11-26:
Hello @jakubhecko,

I shared your issue with our team, and they confirmed that an enhancement will be made. Additionally, after reviewing RM0456 on page 1069, `Table 253 Address alignment cases`, it is recommended that you program 2 bytes and use Status Match with HAL_XSPI_AutoPolling_IT() for RDSR.

Thank you again.
With regards,

KRASTM, 2024-11-26:
ST Internal Reference: 196918

RJMSTM, 2025-01-24:
Fixed in : 007ba8366ad9c870b9df63b50bbcc76cb6788b58