STMicroelectronics/stm32u5xx-hal-driver #22: GPDMA 2D Addressing Repeated Block Configuration overwritten by ADC Byte length calculation

JsGraaf, 2025-06-04:
Line 2520 in stm32u5xx_hal_adc.c:

`hadc->DMA_Handle->LinkedListQueue->Head->LinkRegisters[NODE_CBR1_DEFAULT_OFFSET] = (uint32_t)LengthInBytes;`

This overwrites the repeated block configuration inside of the link register, discarding the functionality.

KRASTM, 2025-06-11:
Hello @JsGraaf,

Thank you for the report.

Could you please share more details and/or your configuration if you encountered a problem with this line of code in your application?

With regards,

JsGraaf, 2025-06-11:
Hello @KRASTM,

I have configured the ADC4 peripheral of the STM32U5A9 with channel 15 of the GPDMA1 peripheral. I've used 2D addressing with 1024 repeated blocks, each sized 4096 uint16_t. I use CubeMX to generate HAL code but my DMA callbacks triggered too soon. Upon investigating the registers of the DMA, I found that the repeated block register wasn't set, only the amount of bytes that needed to be transferred. After looking at the ADC code, the correct parameters are stored in the NodeConfig, but they don't get transferred properly to the LinkRegisters (the 2d addressing parameters don't get set, only the length in bytes). 

Please let me know if you require more specific information or screenshots of CubeMX

JsGraaf, 2025-06-30:
The IOC file is added in .txt format. Below is the section from the hal adc driver which is responsible for setting the LinkRegisters when the DMA is set in standard request mode but with 2D addressing. It shows that the CBR1 register is completely overwritten, discarding any repeated block configuration that is set in the HAL initialization of the DMA. 

```if ((hadc->DMA_Handle->LinkedListQueue != NULL) && (hadc->DMA_Handle->LinkedListQueue->Head != NULL))
          {
            /* Length should be converted to number of bytes */
            if (HAL_DMAEx_List_GetNodeConfig(&node_conf, hadc->DMA_Handle->LinkedListQueue->Head) != HAL_OK)
            {
              return HAL_ERROR;
            }

            /* Length should be converted to number of bytes */
            if (node_conf.Init.SrcDataWidth == DMA_SRC_DATAWIDTH_WORD)
            {
              /* Word -> Bytes */
              LengthInBytes = Length * 4U;
            }
            else if (node_conf.Init.SrcDataWidth == DMA_SRC_DATAWIDTH_HALFWORD)
            {
              /* Halfword -> Bytes */
              LengthInBytes = Length * 2U;
            }
            else /* Bytes */
            {
              /* Same size already expressed in Bytes */
              LengthInBytes = Length;
            }

            hadc->DMA_Handle->LinkedListQueue->Head->LinkRegisters[NODE_CBR1_DEFAULT_OFFSET] = (uint32_t)LengthInBytes;
            hadc->DMA_Handle->LinkedListQueue->Head->LinkRegisters[NODE_CSAR_DEFAULT_OFFSET] =                  \
                (uint32_t)&hadc->Instance->DR;
            hadc->DMA_Handle->LinkedListQueue->Head->LinkRegisters[NODE_CDAR_DEFAULT_OFFSET] = (uint32_t)pData;
            tmp_hal_status = HAL_DMAEx_List_Start_IT(hadc->DMA_Handle);
          }
          else
          {
            tmp_hal_status = HAL_ERROR;
          }
```

[AudioCatV2_SystemCheck.txt](https://github.com/user-attachments/files/20978448/AudioCatV2_SystemCheck.txt)

KRASTM, 2025-07-10:
ST Internal Reference: 213899