STMicroelectronics/stm32u5xx-hal-driver #23: HAL_UART_Receive() blocks if rx Overrun Error occurs

marco-palumbi, 2025-06-23:
When called with Timeout = HAL_MAX_DELAY the function HAL_UART_Receive() blocks if rx Overrun Error occurs.

Looking at the source code:
https://github.com/marco-palumbi/stm32u5xx-hal-driver/blob/b6fc08c2d780e2daf321f2a1d3b51a15c41ec373/Src/stm32u5xx_hal_uart.c#L3476

if `Timeout` is equal to `HAL_MAX_DELAY` both the check for the timeout and the code that resets the `UART_FLAG_ORE` flag are skipped.

Moving the code from line 3484 to 3520 outside the `if (Timeout != HAL_MAX_DELAY)` solves the issue

KRASTM, 2025-06-25:
ST Internal Reference: 212934

KRASTM, 2025-07-07:
Hello @marco-palumbi,

I have checked with our team, and it seems that, when selecting HAL_MAX_DELAY as the timeout parameter, the Timeout is no longer checked and are just waiting for the flag.

With regards,

marco-palumbi, 2025-07-28:
Hello @KRASTM,

the past week I've updated the PR #24 with a small project and instructions on how to replicate the bug.
Did you got time to check it?

regards

KRASTM, 2025-07-30:
Hello @marco-palumbi,

I double checked with our team, and unfortunately it was implemented like that intentionally, `when Timeout is set to HAL_MAX_DELAY, HAL_TIMEOUT will be never returned`, I'm not saying you're wrong, but it was a decision made by our development team.

So, either use your fix locally if it suits your application or use (HAL_MAX_DELAY - 1).

Thank you in advance for your understanding. Please allow me to close the issue.

With regards,